Francklin2/RTKLIB_Touchscreen_GUI #1: Remove sudo in RTKBASE.sh

kikislater, 2016-05-27:
The usage of sudo should be remove.
User could poweroff the pi without typing sudo.

Proposal : 

1) Add $USER to sudoers and create alias in $USER/.bashrc_aliases conf file
2) Use dbus command

dbus command should be tested as it's very more simple. I don't have Raspberry Pi at this time but I could test in a virtual machine inside Qemu

kikislater, 2016-06-10:
Launching RTKBASE with sudo is nt a good idea !

Francklin2, 2016-06-10:
Yes, I will look for a example of dbus command to launch it, is it possible to do a script to add user in the sudoers ?  it could ba include in the dependencies script?

kikislater, 2016-06-10:
I could modify or create a temporary branch to test 
From line 124 in mainwindow.cpp change : 

` process.startDetached("sudo halt");`

to
` process.startDetached("/usr/bin/dbus-send --system --print-reply --dest="org.freedesktop.ConsoleKit" /org/freedesktop/ConsoleKit/Manager org.freedesktop.ConsoleKit.Manager.Stop");`

Francklin2, 2016-06-10:
The sudo was also back because it was needed for access the usb and serial port, they have to be added to the sudoer if I remember

kikislater, 2016-06-10:
Ok there is a workaround also for this but could be more specific for each device because we have to add device to whitelist and each device have a different id. I need raspberrypi to test

Francklin2, 2016-06-10:
The 3 devices used to input/output are : 
/dev/ttyACM0
/dev/ttyAMA0
/dev/ttyUSB0
if it can help you until you have the raspberry

Francklin2, 2016-06-19:
Atfer test the sudoseemed to be not needed on the pi to serial and usb ports, so it can be removed for now

kikislater, 2016-06-19:
Nice ! I ever found that pi user have special administrative rights on raspbian ...

Francklin2, 2016-06-25:
Yes so a sudo is not needed anymore, we can close this issue for the moment

closed 2016-06-25 by Francklin2