Francklin2/RTKLIB_Touchscreen_GUI #2: Create simple script to build RTKBASE

kikislater, 2016-05-27:
RTKBASE is built with several command like : 
- Build static rtklib (there is a file problem : it's not executable => don't know if I could change the rights of this file under github)
- Then building RTKBASE

It's more simple to add one command to do that like run_build.sh or other name. Script below : 

`#!/bin/bash
cd lib/rtklib
chmod +x make_library.sh
sh make_library.sh
cd ../../
# Cleaning
make clean
# Builld RTKBASE
qmake
make`

Francklin2, 2016-05-27:
Thanks for the script, it solve the rights mistake in the files (I tried to update the rights before uploading with chown  pi:pi -R /home/pi/RTKBASE but i'm not sure if the rights are still good after transfering from the raspberry to desktop pc and uploading here, so this script is a good solution to prevent files rights issues

kikislater, 2016-05-27:
Rights problems are on make_library.sh under rtklib not on RTKBASE :)

kikislater, 2016-05-27:
So script solve, just need to modify wiki. I'll do

closed 2016-05-27 by kikislater