Francklin2/RTKLIB_Touchscreen_GUI #3: Starting versionning

kikislater, 2016-05-28:
Hi Francklin,

I created a new branch which is the copy of the actual revision.
Please start development or modifying files in master branch.

What do you think about versionning and numerotation ? Starting at 1.0 is a good choice ?

Francklin2, 2016-05-30:
Thank's a V1.0 is ok because the software is working, I will make the changes in the master branch now.
I noticed that  the radiobuttons in the rover option page use to select the input/output stream are not functionnal (only the button on the ui ), I note it in the wiki (base mode) and I will start a new issue for it.
.

kikislater, 2016-05-30:
Ok, so I close this one. Thanks for the feedback

closed 2016-05-30 by kikislater