Francklin2/RTKLIB_Touchscreen_GUI #5: Selection of auto or manual base position is not working

Francklin2, 2016-05-31:
Only added the radiobuttion to optionsstr2str.ui . I have to put a if condition to select position file  in affichestr2str.cpp

Francklin2, 2016-05-31:
I put a filename option in affichestr2str to switch between the 2 position filenames, but it dont work I"m not sure the PositionMode variable from optionsstr2str.cpp  is  shared

Francklin2, 2016-06-02:
Changed the déclaration of the positionMode définition in optionsstr2str.h to be accessible in affichestr2str.cpp, It was déclared like this:
in optionstr2str.cpp:
`#include "optionsstr2str.h"
#include "ui_optionsstr2str.h"
#include "affichestr2str.h"
#include <QTextStream>
#include <QFile>

 int PositionMode;
`
in optionsstr2str.h:
`#ifndef OPTIONSSTR2STR_H
#define OPTIONSSTR2STR_H

#include <QDialog>
extern int PositionMode;`
Now there is no more error "PositionMode was not declared" in affichePositionMode.cpp in the compilator, so I guess the variable PositionMode is accessible by affichestr2str.cpp.
 in affichestr2str.cpp I tried a if/else condition on PositionMode to select filename but in that case , only the first condition is token (auto mode for PositionMode=1),.

`{
    args=arga;
    ui->setupUi(this);
    /*Displays in FullScreen*/
    this->setWindowFlags(Qt::Window);
    this->showFullScreen();

    /*pick up Base Position from last recorded in rover mode*/

     QString filenamebase;

{

  if (PositionMode = 1) {filenamebase = "sauvegardepourbase.txt";
  } else {filenamebase = "sauvegardepourbaseManual.txt";}

}

    int i=0;
    QStringList list;
    QFile fichierlastpositionforbase(filenamebase);
    fichierlastpositionforbase.open(QIODevice::ReadOnly | QIODevice::Text);
    QTextStream flux(&fichierlastpositionforbase);
    QString ligne;
    while(! flux.atEnd())
    {
        ligne = flux.readLine();
        list<<ligne;
        i=i+1;
    }
    fichierlastpositionforbase.close();
`

If I put a 2 if like this, it will only take the second condition (manual mode for PositionMode=2).
`
{

`if (PositionMode = 1) {filenamebase = "sauvegardepourbase.txt";}

 if (PositionMode = 2) {filenamebase = "sauvegardepourbaseManual.txt";}`

 }
`

Francklin2, 2016-06-02:
I found the mistake I have to put 

`if (PositionMode == 1)

Instead of 

`if (PositionMode = 1)

kikislater, 2016-06-02:
Eheh, so basic but you found it !

Francklin2, 2016-06-02:
Yes it was quite basic but  sometime  the harder to see :)))

It"s functionnal now so I can close this issue (just doing a few graphics modifications)

closed 2016-06-02 by Francklin2