Francklin2/RTKLIB_Touchscreen_GUI #7: Help button close application

kikislater, 2016-06-01:
Is it a functionnality to close the software when clicking on Help button ?
I think it's not a good idea. Help come in foreground, when we close the pdf viewer launched by help button we should return to the software. Tell me what you do you think.

The workaround is very simple. Actually we have this code in mainwindow.cpp : 
void MainWindow::OuvreHelp()
```
{
    QDesktopServices::openUrl(QUrl("./doc/manual_RTKBASE.pdf"));
    //this->hide();
    qApp->quit();
}
```
We just have to comment the line 132 like this : 
void MainWindow::OuvreHelp()
```
{
    QDesktopServices::openUrl(QUrl("./doc/manual_RTKBASE.pdf"));
    //this->hide();
    //qApp->quit();
}
```

Francklin2, 2016-06-01:
Yes it's a good idea to remove this, i noticed too that it was not helpfull to close the app like this

kikislater, 2016-06-01:
Ok update and closed

closed 2016-06-01 by kikislater