Francklin2/RTKLIB_Touchscreen_GUI #8: Base position used by str2str server is a fixed value

Francklin2, 2016-06-02:
I saw in the code that the  -p  option send to str2str is a fixed value for testing but not the position from the base position files.the default button send this string :+1: 

> arga={"carlepremierargtoujorsleprog","-in","serial://ttyACM0:115200:8:n:1:#ubx","-out","serial://ttyUSB0:38400:8:n:1:#RTCM3","-msg","1004,1019,1012,1020,1006,1008","

The part `-p","48.2","12.2","120.23"}; should have variables instead of fixed value`

kikislater, 2016-06-02:
Ah yes ! I saw that many times in files ...  That's a big problem

Francklin2, 2016-06-03:
So perhaps it's just a wrong  display in the GUI , I saw that the position of the base displayer un base stream status seemed to be taken fram the str2str serrunning server. The display code is there (affichestr2str.cpp):

>void AfficheStr2str::recupdonneesStr2str(QStringList i)
{
    {
         if (PositionMode==1) { ui->Msg1lineEdit->setText(QString("AUTOMATIC BASE POSITION picked up from last rover mode"));}
         if (PositionMode==2) { ui->Msg1lineEdit->setText(QString("MANUAL BASE POSITION entered by user"));}
    }
    ui->Msg2lineEdit->setText(QString("LAT = %1 deg   LONG = %2 deg   ALTI= %3 m").arg(m_str1).arg(m_str2).arg(m_str3));

I wil try to see which base position is receive by the rover to check what is really send

Francklin2, 2016-06-04:
Finaly I have to put a empty string in arga: arga={""} and put the args on the command string for str2str (or supposed to be I"m still not sure of that point), like this I  can add the wriables for base position.
The correct base position is now displayed in the "CURRENT STR2STR OPTIONS" screen , so that part is ok now but I see that the RTCM strings send by the base are not readable. I think I we can close this issue a open a new one for the rtcm stream

Francklin2, 2016-06-05:
I could check that the string used to launch str2str and displayed in GUI  is the same and really used by the library, so I can close this issue (I found another one in the transmited base position, thi will be the nest issue)

closed 2016-06-05 by Francklin2