Francklin2/RTKLIB_Touchscreen_GUI #9: RTCM data send by the base mode  is receiced but not readable

Francklin2, 2016-06-04:
Now yhe str2str server is launched by strings so the correct base position should be inputed, a stream is send by the base and received by the rover but the stream looks unreable by the rover  (tried with the microsoft app and same résult).
I think the string used to launch str2str is wrong,  on the display inn CURRENT STR2STR OPTIONS we see that the each blank space return to next line, I can be only the display but I"m not sure...
On the old base the string used for messages was :+1: 
> -msg 1004,1019,1012,1020,1006,1008
and in our string the is:

> -msg "1004,1019,1012,1020,1006,1008"
I dont know how to add the "" in the string correctly, it alway think it's a new string !

kikislater, 2016-06-05:
Seems rtklib static lib doesn't interact with rtkbase

Francklin2, 2016-06-05:
AAArgh I found the issue, so simple but I spend a day on it hihi...the baudrate in the string was not the correct one for my radio , it was 38400 and should be 57600 for me, but rtklib dont send any error in this case (stream ok) the rtcm is just unreadable.
So check well the radio modem speed and the speed have to be enter in the args string (affichestr2str.cpp).
I can close this issue , the good thing is thaht I could check wich string is used to launch str2str

closed 2016-06-05 by Francklin2