Francklin2/RTKLIB_Touchscreen_GUI #15: Can't compil

Stefal, 2016-06-10:
Hi !
I can't compil RTKBASE on my Raspberry Pi 2. Here is the logs:

[logs_pi2.zip](https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/files/309237/logs_pi2.zip)

kikislater, 2016-06-10:
Sorry, need to investigate ... 
Which system do you have on your Pi ?

Stefal, 2016-06-10:
I have the raspbian image linked in the readme.md:
https://drive.google.com/file/d/0B2SuLGFxizpVMzRtODVkN0tZVjA/view?usp=sharing
Raspbian 7 (Wheezy)

Francklin2, 2016-06-10:
I tried on the same image 
It load some dependencies and compile rtklib  , but the building of rtkbase still dont work ( send you the logs).
I can try to lauch the qtcreator install to see which library are required for the raspberry (if qtcreator is installed, the script works)

kikislater, 2016-06-10:
Yes I think a libqt4-xxx is missing

kikislater, 2016-06-10:
I think it's this one : libqt4-dev-bin

Francklin2, 2016-06-10:
I try to install these dependencies but tey where already installed but not upgraded, i try a apt-get update/upgrade to be sure , I send tou the return of the gcc -v by mail

Francklin2, 2016-06-10:
> Aucun virus trouvé dans ce message.
> Analyse effectuée par AVG - www.avg.com <http://www.avg.com>
> Version: 2016.0.7639 / Base de données virale: 4598/12393 - Date:
> 10/06/2016

Salut

si tu as envie de tester l'appli, tu peux installer qtcreator pour 
compiler, mais du coup on ne sait pas laquelle est necessaire pour notre 
script de dependances, tu peux l''installer avec apt-get install qtcreator

Francklin2, 2016-06-17:
I followed the  install instruction of the wiki to copy files properly, so with qtcreator installed it compile with the script or Qtcreator.
If some are still stuck with compiling the program ,This is a link to a pré-compiled version for pi , tell me if it works on your pi

https://www.dropbox.com/s/1twbh82jf1wp8gd/RTKBASE.ZIP?dl=0

kikislater, 2016-06-17:
Ok so we can add qtcreator as a dependency for a temporary workaround ? 
After qtcreator installed, does the script build_rtkabse.sh is working ?

Francklin2, 2016-06-17:
To confirm that with the disk image for touchscreen I will reinstall it this weekend (I play a lot with this install...) but it should work with qtcreator as a dependency

Francklin2, 2016-06-18:
With only QTcreator installed, it still not compile with the build script, In qtcreator I have to configure first the gcc toolchain before compiling(didn't do it yet), Perhaps it was the problem, This evening I  will configure Qtcreator toolchain and compiler to see if the build script workafter that

kikislater, 2016-06-18:
Ok !
Did you notice that I have modified the script with commenting this line and adding comment to test with pi2 ? : 
#LC_ALL=C #>>>> TEST MODIF pour build pi2

Did you try with this ?

Francklin2, 2016-06-18:
Yes I used that version,I'm trying to configure tcreator and saw in this tuto , perhaps you will find something interesting in there 
https://wiki.qt.io/Apt-get_Qt4_on_the_Raspberry_Pi

kikislater, 2016-06-18:
Yes thank you, I saw that :  
`Then set compiler path : /usr/bin/arm-linux-gnueabihf-gcc-4.6 `

That's what I asked last time by mail but you send me only your desktop configuration : 
Tell me what you see here : 
In Qtcreator go to Tools -> Options..-> Build & Run -> Compilers

kikislater, 2016-06-18:
I saw also this dependency : qt4-dev-tools 
And there are lot of dependies with this package

Francklin2, 2016-06-18:
Yes I install them all with no success, I also tried to compile with qtcreator and get the same error, 
I saw this in a read me file : 
Using QCustomPlot as shared library .so/.dll
Using a shared library means to not include the .h/.cpp file into your project, but linking with an external qcustomplot.so (GNU/Linux) or qcustomplot.dll (MSWindows) file. QCustomPlot is ready to be built as a shared library by setting the compiler define QCUSTOMPLOT_COMPILE_LIBRARY. To use the shared library in your application, set the define QCUSTOMPLOT_USE_LIBRARY before including the QCustomPlot header.

The sharedlib package in the download section provides two projects that demonstrate this: one compiles the shared QCustomPlot library and the other uses the shared library. This should quickly get you started using QCustomPlot as a shared library.

http://www.qcustomplot.com/index.php/tutorials/basicplotting

Francklin2, 2016-06-19:
Finaly Until we found the trick, I upload the Raspbian image with the working Qt4, I update the reas me and the wiki withe the new link:

https://drive.google.com/file/d/0B5xmVAi4jATOcXE3a3BWS0VDUUk/view?usp=sharing

kikislater, 2016-06-19:
So what there is in the iso @Francklin2  ? 
Raspbian image only or Raspbian and RTKlib touchscreen ? Does the previous image a bad iso ?

Francklin2, 2016-06-19:
The previous iso was the Raspian image with touchscreen only we use before adding Qt4, but I don't remenber how we did  to make it compile  so I put a iso of the working one we get at the end (the one where the building script and Qt4 works)

kikislater, 2016-06-19:
Ok !

kikislater, 2016-10-01:
sudo apt-get install libqt4-dev
sudo apt-get install libqt4-core
sudo apt-get install libqt4-gui
sudo apt-get install libqt4-xml
sudo apt-get install libqt4-opengl

Corrected dependencies above

closed 2016-10-01 by kikislater