Francklin2/RTKLIB_Touchscreen_GUI #16: It wasn't possible to execute ./RTKBASE

alejoast, 2016-06-15:
Hi.
I managed to successfully execute the ./build_rtkbase.sh after using the "chmod u+x build_rtkbase" command because it had not execution privileges. After that, I was willing to execute the RTKBASE.sh but it says that the command doesn't exist. I mannualy execute it, but the terminal just pop-up for like 0.5 sec and is closed again.
I'm using a Raspberry Pi 2 B+. I don't know what else to do.

kikislater, 2016-06-16:
Hi,
Yes this is because we have trouble at this moment to build RTKBASE. You should wait a couple of days 
Same as https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/issues/15.
There is a prebuilt package here : https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/releases Please use this one but there is some bugs

Francklin2, 2016-06-16:
I will put a pré compiled software fol der in thé github un il de fond the  compilation  problem

alejoast, 2016-06-16:
Ok. Thank you very much. If you need any logs or something, just let me know. I'll be aware.

kikislater, 2016-06-16:
I need a raspberry pi that I don't have today, but it's on the way ...

Francklin2, 2016-06-17:
This is a link to a pré-compiled version for pi , tell me if it works on your pi

https://www.dropbox.com/s/1twbh82jf1wp8gd/RTKBASE.ZIP?dl=0

alejoast, 2016-06-17:
Hi Francklin, it worked on my Raspberry Pi 2 B+. I didn't see the full GUI as shown in the pdf manual. I think this pre-compiled version is an old one, I'm right?. Because, it was not possible for me to select any of the configuration files, change the Base output or Base input to TCPClient for example. Thanks for this pre-compiled version, I'll be waiting for the last version.

Francklin2, 2016-06-17:
To edit the configuration files for the rover mode you have to exit the gui and edit the ubx files in the ConfFile folder of RTKbase , in these files you will find the input/output parameters, I forget to update the pdf manual so it's better to look in the wiki manual for the moment

alejoast, 2016-09-30:
Hi. I want to update the state of this issue. I've achieve to compile the RTKBASE in a Raspberry Pi 3 with Touchscreen drivers.
I only needed to use the following commands in the terminal:
`sudo apt-get install libqt4-dev
sudo apt-get install libqt4-core
sudo apt-get install libqt4-gui
sudo apt-get install libqt4-xml
sudo apt-get install libqt4-opengl

sudo apt-get upgrade

git clone https://github.com/Francklin2/RTKLIB_Touchscreen_GUI

cd  RTKLIB_Touchscreen_GUI/RTKBASE/

chmod u+x build_rtkbase.sh

./build_rtkbase.sh

./RTKBASE`

Thank you very much for your help.
I think now you can say that this RTKLIB GUI has been proved to work also in a RasPi3.

kikislater, 2016-10-01:
Thanks for the feed

kikislater, 2016-10-01:
Added to wiki

closed 2016-10-01 by kikislater