Francklin2/RTKLIB_Touchscreen_GUI #17: Add how to autolaunch software to wiki

kikislater, 2016-06-16:
Very simple : 

**1) Add software to rc.local** 
sed -i -e '$i \cd  /home/pi/RTKLIB_Touchscreen_GUI/RTKBASE/ && RTKBASE\n' rc.local

**2) Use system root LXDE autostart :** 
Run : 
`sudo nano /etc/xdg/lxsession/LXDE-pi/autostart`
Add this : 
```
@/home/pi/RTKLIB_Touchscreen_GUI/RTKBASE/
@RTKBASE
```
3) Use user LXDE autostart : 
Run : 
`nano ~/.config/lxsession/LXDE/autostart`
Add this :
```
@/home/pi/RTKLIB_Touchscreen_GUI/RTKBASE/
@RTKBASE
```

**4) Add desktop entry to user config :** 
Run : 
`nano ~/.config/autostart/rtkbase`
Copy paste entry below : 
```
[Desktop Entry]
Name=RTKBASE
Type=Application
Comment=RTKLIB Touchscreen : graphic interface to use GNSS.
Exec=/home/pi/RTKLIB_Touchscreen_GUI/RTKBASE/RTKBASE
```
Make it executable : 
`chmod +x ~/.config/autostart/rtkbase`

Best way for me is option 4 because I do not use LXDE

Source : http://www.raspberrypi-spy.co.uk/2014/05/how-to-autostart-apps-in-rasbian-lxde-desktop/ and my own knowledge

Francklin2, 2016-06-17:
Thank's I will add it to the wiki and the readme.md

kikislater, 2016-06-17:
Need to adapt the path. May be I am wrong with /home/pi/RTKLIB_Touchscreen_GUI/RTKBASE/RTKBASE
I give you a feedback later

Francklin2, 2016-06-17:
I just add it in the wiki and readme, I think it's 
/home/pi/RTKLIB_Touchscreen_GUI/RTKBASE/
But I will check it tonight

Francklin2, 2016-06-25:
Added to the readmd.md and the wiki, I close this one

closed 2016-06-25 by Francklin2