Francklin2/RTKLIB_Touchscreen_GUI #18: Modifying Readme.md

kikislater, 2016-06-19:
I have modified the readme file.

If you have suggestion, please send your message here.

Francklin2, 2016-06-19:
Great ,it's looks better with the miage now, the mink with your precision tests on Rtklib précision gives a lot of answers  about the résults we cpuld have

Francklin2, 2016-06-25:
I added a link to a Raspbian iso with pré-installed RTKbase (latest master) it has Qtcreator and compiler installed so it can be easyly updated

kikislater, 2016-06-26:
Nice, so user don't need to build software itself

kikislater, 2016-06-29:
Adding some tests results

Francklin2, 2016-07-05:
Thank's for the test resuluts, I changed the pictures of the RTKbase (putted nices screw on it and took a real camera and not my phone ...)

kikislater, 2016-07-05:
Nice ! Not so much time at this moment, but I could be able to test it on a drone ...

closed 2020-01-20 by kikislater