Francklin2/RTKLIB_Touchscreen_GUI #19: Proposal antenna for rover

kikislater, 2016-06-20:
Hi Francklin, 

I found this antenna which seems to be the same as the implemented ebee one or heliceo or drobit system => 25g 

> Antenna Spec
> Frequency Range 	GPS  L1/L2
> GLONASS  L1/L2
> BDS B1/B2
> Impedance 	50ohm
> Polarization 	RHCP
> Axial Ratio 	≤3dB
> Azimuth Coverage 	360°
> Output VSWR 	≤2.0
> Peak Gain 	2dBi
> LNA Spec
> LNA Gain 	33±2dB
> Noise Figure 	≤2dB
> Output VSWR 	≤2.0
> Operation Voltage 	3~12VDC
> Operation Current 	≤55mA
> Ripple 	±2dB
> Mechanical Spec
> Dimension 	¢27.5*59mm
> Weight 	≤25g
> Connector 	SMA
> Environment Spec
> Operating Temp 	-40℃～+70℃
> Storage Temp 	-40℃～+70℃
> Humidity 	95% No-condensing
> 

http://gpsworld.com/harxon-offers-new-gnss-helix-antenna/

http://www.harxon.com/products-detail.php?ProId=83

kikislater, 2016-06-20:
An idea about price is ~250€ without freight cost

Francklin2, 2016-06-20:
A bit expensive but it will be a nice one for the rover (25grams) or a good alternative for the talysman, we can add it to the shopping list

Francklin2, 2016-06-21:
Around the same price I saw this new one at Csgshop, not really a antenna because it got the intégrated M8T , it could be a middle price solution for the rover side (there is cheaper ones aroud 90$ at Drotek or Csgshop)
http://www.csgshop.com/product.php?id_product=222

http://www.csgshop.com/img/p/222-518-thickbox.jpg

kikislater, 2016-06-23:
With this size and weight of 25g, there are some different and a little bit cheaper for L1 ~around 180$

Model  	Frequeny
HX-CH4601A 	GPS L1/L2，GLONASS L1/L2
HX-CH6601A 	GPS L1/L2，GLONASS L1/L2 BDS B1/B2
HX-CH3602A、HX-CH3603A、HX-CH3604A 	GPS L1、BDS B1、GLONASS L1

Of course your M8T you list is cheaper but only L1

Francklin2, 2016-06-25:
We can propose a list of compatible GNSS in different prices range, I only search for the low cost side for the moment (Ublox,Skytrack or NVS) 
http://navspark.mybigcommerce.com/ns-raw-carrier-phase-raw-measurement-output-gps-receiver/
http://www.navspark.mybigcommerce.com/ns-hp-rtk-capable-gnss-receiver/

kikislater, 2016-06-28:
According to ublox specs we could have 10Hz with setting only GPS or GLONASS in M8T 
P7/35 => 
https://www.u-blox.com/sites/default/files/NEO-LEA-M8T-FW3_DataSheet_(UBX-15025193).pdf

May be usefull for rover on a drone

Francklin2, 2016-06-28:
Yes the M8T can run at 10hz, you have to set it in Ucenter, to change the refresh rate in hz, you have to change the timing in ms to 200ms instead of 500ms., over 5hz some values become red but it works i will add how to setup the M8T in the wiki

kikislater, 2016-06-29:
Before known test, I prefer using 10Hz on a drone : more reliable

kikislater, 2016-07-10:
Francklin, did you see the latest version : 
https://pixhawk.org/dev/rtk-basestation-touch-tft

Francklin2, 2016-07-11:
Yes I saw it, I have build the first version without screen and we started our project  in the same time, we can tell him that our project is released too., I didnt try his software yet because It need the correct touchscreen . I will try to download it to see how he extract the info from rtkrcv.
For the rover part I noticed that the Drotel Smartnav is not too expensive (288€)