Francklin2/RTKLIB_Touchscreen_GUI #20: Add a custom spash screen at boot

Francklin2, 2016-06-25:
On my Raspbian image I add a custom splash screen and removed boot messages,it works but I'm not sure it's the best method to do this. Here is what I did:

- Install fbi with sudo apt-get install fbi
- Add splash image to etc folder (splash.png)
-Create a asplashscreen file in etc/init.d/asplashscreen and put the script found in this tuto:
http://www.raspberry-projects.com/pi/pi-operating-systems/raspbian/custom-boot-up-screen
- Make the script executable and install it for init mode:
sudo chmod a+x /etc/init.d/asplashscreen
sudo insserv /etc/init.d/asplashscreen

- Edit /boot/cmdline.txt to remove text message and add loglevel=3 at the end of file and change console=tty1 to console=tty3

kikislater, 2016-06-25:
Splash screen on boot when Raspian is loading ?

Francklin2, 2016-06-25:
Yes it was the idea,I dont know if it was the best way to do it

kikislater, 2016-06-28:
I don't know raspberry pi system but seems to be good

Francklin2, 2016-07-02:
It will be more easy for you to see that when you will have you raspberry, it's working well for the moment so I  upload the splash image in the IMG folder and I wil add the instruction for splash screen in the wiki and readme.md

Francklin2, 2016-07-05:
I put all files for splashscreen in a separate folder with a install manual

closed 2017-04-10 by Francklin2