Francklin2/RTKLIB_Touchscreen_GUI #21: Add latest rtklib on master branch

kikislater, 2016-06-25:
Due to this bug : https://github.com/tomojitakasu/RTKLIB/issues/174#issuecomment-228187487
I would be nice to have the latest rtklib version on master branch

Francklin2, 2016-06-25:
Yes good idea and it could add Galileo compatibility, folders have been changed since 2.42 , did you see wich files have to be updated ?

kikislater, 2016-06-25:
Yes, but I have to test...

Francklin2, 2016-06-25:
Ok I found back the sources files (simply in src folder) so I just put the 2.43 /src folder in our /rtklib folder, I have test compilation on desktop PC (ubuntu)  and it works, still have to be tested on Rpi2 but it should be ok

kikislater, 2016-06-26:
Very usefull upgrade. Thanks

kikislater, 2016-06-26:
Reopened, need to update rtklib.h to specify version but don't understand why it's not done on rtklib main stream

Francklin2, 2016-06-26:
I test it on a Raspberry pi and it seemed to compile normaly (but I didn't check the logs, and didn't test it with a M8T plugged)I still have to update my (complete) rtkbase to see if everything works after RTKlib update

Francklin2, 2016-06-28:
I use the new version of rtklib for the vidéo and it worked , so we jsut have to add the new version in the list

kikislater, 2016-06-28:
Ok, I think we have to keep revision 1 but could add subversion like 1.1 or 1.2 or other
Master is master and have to keep master with latest improvements and testing. It's my opinion, yours could be different of course :D

Francklin2, 2016-07-02:
Do you know  how to go back before I update rtklib and do a release of it ? I'm a bit afraid to do a mistake.on the master branch

kikislater, 2016-07-02:
Going back to 2.4.2 ?

Francklin2, 2016-07-05:
I'm not sure , if it works well in 2.43 we can release it in 2.43 or do you think it's more safe to do a release in 2.42 until we do more tests

kikislater, 2016-07-05:
There is some bugs in 2.42 with ublox M8x, for me it's better to have upstream

Francklin2, 2016-07-05:
Ok so we can keep 2.43 and do a release while it's stable

kikislater, 2016-07-05:
I saw 2.43 b13 at this time ... I have to check but seems to concern M8P only

Francklin2, 2017-08-26:
I updated the master branch with 2.4.3b28 and create release under Tag v1.1 for orignal release with 2.4.2p11 and release Tag v1.1.1 with 2.4.3b28

Francklin2, 2017-09-09:
As 2.4.3b28 still doesn't  build well with rtkbase, I put back 2.4.2p12 in master

Francklin2, 2017-09-13:
In 2.4.3 I saw that PPP is not possible with a L1 single frequency like the Ublox M8T (Issue #305 in RTKlib), I don't know yet if it's for all version in 2.4.3, we can stay in 2.4.2p12 for a stable version.
I was able to made a working 2.4.3b29 build, where should it be uploaded ? the master branch or another branch or fork ?

kikislater, 2017-09-13:
Why not using branch ? 
So to be testing, I suggest name of branch or tag with rtklib version. In that case you could find in the future what you make like here : https://github.com/emlid/RTKLIB

Why you use PPP ?

Francklin2, 2017-09-13:
I tried PPP to get base position if you have internet connexion for corrections(and just to try if it worked), in the base preset I will add a data logging setting to make it more easy to use.
I will upload version 2.4.3 in a branch called rtkbase_2.4.3 , we have to look if the emlid 2.4.3 is able to do PPP in L1

kikislater, 2017-09-13:
I have reach, it works in ppp. But I use them most often in ppk so I capture data in static.
Higher fixes in ppk => https://rtklibexplorer.wordpress.com/2017/08/21/ppk-vs-rtk-a-look-at-rtklib-for-post-processing-solutions/

Here you have good reading about ppp : https://community.emlid.com/t/24hr-reach-collect-nrcan-ppp-results/5658 
I had make real life test with known coordinates. even if you log long period : more than 24h, it will not improve location. You will stay in 20-30cm confidence. If you want/need more, a dual frequency is needed to make difference between L1 and L2 and improve position. I have a trimble R8, I could compare...

Francklin2, 2017-09-17:
Thank you for the links, I will study that to see how they have ppp with Ublox in 2.4.3,  I still have to translate the ENGS test before put them in the wiki but they have quite the same results with 20cm at the best. They say they have some offset difference with a know point, can you make the same test with your trimble to see if we have a correct position ?.
To update the libray in 2.4.3 I compared the original 2.4.2p12 and the one from rtkbase, I take the modified parts and put them in 2.4.3. the files to be modified in rtkbase are:
rtkrcv.c
str2str.c
vt.c
vt.h
I have to check again if 2.4.2p12 was done correctly but it looks good.
I can't create a branch with 2.4.3, if I add the folder in the branch and make a commit,git retun a error> I have to put it in trunk before, so I made a release of the actual master (v1.3) to save it . I don't know well github, is it possible to create a 2.4.3 branch directly or I have to put it in master and move it after?

Francklin2, 2017-09-17:
I create a RTKbase-2.4.3 branch, it build so I can close the build issue. About the unused variable error, some of them could be normal because some variable are used in rtknavi and not in rtkrcv

Francklin2, 2018-04-28:
Sources were updated to 2.4.2p13, 2.4.2p13 was merged with 2.43 in january 2018 and latest updates on rtklib  were done on 2.4.2p13 (march 2.18)