Francklin2/RTKLIB_Touchscreen_GUI #23: doesn't work with Navspark NS-HP

Stefal, 2016-08-22:
Hi !

I tested with a NS-HP, but it doesn't work.
If I change the settings in the conf files, it always wait for ubx data in the status window.

Francklin2, 2016-08-22:
Hi Stefal
on the conf line
inpstr1-format     =7       # (0:rtcm2,1:rtcm3,2:oem4,3:oem3,4:ubx,5:ss2,6:hemis,7:skytraq,8:sp3)

Did you tried with
inpstr1-format     =skytraq
and
inpstr1-format     =7
I suppose you port setting and the NS-HP are properly configured

Stefal, 2016-08-23:
Oups !  I mixed up the syntax between str2str and rtkrcv

inpstr1-format = stq <----- NOK  (my mistake)
inpstr1-format = 7 <----- NOK (strange.... perhaps because it's an ubx file)
**inpstr1-format = Skytraq <----- OK**

closed 2017-02-12 by Stefal