Francklin2/RTKLIB_Touchscreen_GUI #24: Raspberyy pi 3 and pre installed

Dimitrikakog, 2017-02-06:
I use pre installed image in pi 3 but it not boot
I have the same problem if i use the firmware update in manual install.

Francklin2, 2017-04-10:
Hello
The pré-compiled image was made for a rasperry pi 2 in the 32 bit version of Raspian, it's not compatible with the  raspberry pi 3 in 64 bit. I will try to release a version for the RPI3 this summer or before.
For the moment you have to use a RPI2 or install the drivers for the waveshare touchscreen  and QT on a Raspian for RPI     then compile RTKlib GUI with QT

claudio00033, 2017-04-25:
Bonjour,
Maintenant je comprend pourquoi, je rencontre le même problème que Dimitrikalog avec l'image préinstallée sur un PI3. 
J'ai essayé de compiler RTKLIB GUI avec QT, mais j'ai une erreur de compilation 'class Ui: OptionsStr2str' has no member named 'radioButtonPositionAuto'. En fait, il y a toute une série d'erreurs (environ 80) 'class Ui: OptionsStr2str' has no member named XXXX avec différent XXXX.  Autrement, je n'ai pas relevé d'erreur dans la configuration de QT

Qu'elle peut être la cause de ce problème?

foxittt, 2017-04-29:
Hello. 
Describe how to install the drivers for the waveshare touchscreen and QT on a Raspian for RPI 3 and to compile RTKlib GUI with QT.

Francklin2, 2017-08-15:
For compilation on a Pi3 you nedd to install library before, look in the wiki or try this:
sudo apt-get install --assume-yes libxtst-dev build-essential libqt4-dev qt4-qmake libqt4-core libqt4-gui libqt4-xml libqt4-opengl 

Then run the build script, it should work. I will release a PI3 pre installed image soon

Francklin2, 2017-08-26:
You can find a tuto to install the driver for touchscreen are here :+1: http://www.waveshare.com/wiki/5inch_HDMI_LCD
A preinstalled iso for raspberry PI 3 was uploaded, the links are on the wiki/install

closed 2017-08-26 by Francklin2