Francklin2/RTKLIB_Touchscreen_GUI #29: RTKBASE does not finish building

idognation, 2017-09-01:
I am running Linux 8.0 (jessie) on RPi 3 and I ran into an issue where the RTKBASE does not finish building. I followed the instructions on:
https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/wiki/1---Installation
Followed the part Installation on Raspberry Pi 3 
It gets stuck at the ./build_rtkbase.sh when its building the RTKBASE
[image]

I super appreciate any help I can get on this. Thank you.

kikislater, 2017-09-01:
There is problem with latest uploaded push. It's about moving from rtklib 2.4.2 to 2.4.3
Hope Francklin will help ...

kikislater, 2017-09-01:
According to this commit https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/commit/9e7026ff6cf63ebf89efc1738aa9b05f83d977ce

I updated build script. Try it please

Francklin2, 2017-09-01:
On the Raspberry PI 3 with Raspbian Jessie , I had to run manually the library install, try to run this before the build script:
sudo apt-get install --assume-yes libxtst-dev build-essential libqt4-dev qt4-qmake libqt4-core libqt4-gui  libqt4-xml libqt4-opengl 

If you dont have a ./RTKBASE at the end try to run
make clean
qmake
make

I have to test with a clean raspbian jessie if a special build script is needed for the Pi 3

Francklin2, 2017-09-02:
I tryed with thr new build script on a clean raspbian Jessie and it works well getting the needed library and I get a ./RTKBASE file .
I put the build logs if you want to check , I open a new issue for the 2.4.3 errors on build 
[logs.zip](https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/files/1272526/logs.zip)

Francklin2, 2017-09-17:
The new script build well on Pi 3 so we can close this issue

closed 2017-09-17 by Francklin2