Francklin2/RTKLIB_Touchscreen_GUI #31: RTKbase does not build with RTKlib 2.4.3b29

Francklin2, 2017-09-04:
I was not able to build RTKbase with RTKlib 2.4.3b29, the library build with some errors like in 2.4.3b28, we get a fatal build error on RTKbase. rtkrcv.c and str2str.c have to be updated, first try with just adding the missing new values gives a running software with bugs: base mode crash we we stop the base and PPP can't get a fix.  So I actually try to edit  the original 2.4.3b29 rtkrcv and str2str with RTKbase modifications.
I still cannot build yet but most of the build errors are gone on rtkrcv.c

kikislater, 2017-09-04:
IMO better is to keep 2.4.2 as stable release. 2.4.3 need to look more in depth what is blocking. Have to check others using rtklib like reach or others what they do do make it working ...

Francklin2, 2017-09-04:
I agree , I also seen that the conf files have change too, so it's safer to stay on 2.4.2 for the moment.
I have to re-write the configuration file part, now the use the same file for rtknavi and rtkrcv.
To take off the "unused values error" I have to add them in the affichesolution.ccp I think .
To update i added rtkrcv.c str2str.c rtklib.h vt.c and vt.h from RTKlib 2.4.3 then add the modified parts from working 2.4.2 , I missed a declaration for vt thats why it still dont  build.

kikislater, 2017-09-05:
Pi3 has been proven and it is common in use nowaday, why not building a stable version with 2.4.2 for pi3 ? 
I have one for GIS purpose, I may try !

Francklin2, 2017-09-05:
Yes we can keep the stable master version in 2.4.2 , the new beta version should be stable on 2.4.2 now, I can make iso image for raspberry pi 3 for the master and beta  this weekend.  Nice to hear that you have a Pi3 for the tests.
So I have to  put back 2.4.2p12 in the master and the beta (and check if it's ok with the beta)

kikislater, 2017-09-05:
don't put 2.4.2 p12 in master. There is already a stable branch, finalize stable process on pi2 and 3 and it should be more convenient for users. 
Then keep master on master = test and non stable release. Keep users and visitors updated on front page and wiki. There is possibility to make control over builds but don't remember where. 

It just my point of view, you could not be agree of course :D

Francklin2, 2017-09-07:
I take a look in the RTKlib version in the github and see that I made a lot of mix in the versions...so in the master and beta 1.1 there is 2.4.2p12 , in the tag versions (( wanted to do a release but it create a tag) release 1.0 and V 1.1 are in 2.42p11, v 1.1.1 is in 2.4.3b28 (so I have to remove it).
I noticed in 2.4.2p12 that there is already some "variable set but unused" errors due to support of new GNSS in rtklib and updates in ublox and rtcm . but rtkbase build and works well.
Do you think we should update now the master in 2.4.3 now , or try to take off the builds errors in 2.4.2 and let the 2.42  in master for the moment ?  (the master build well on pi 3 with the build script you updated)

Francklin2, 2017-09-09:
I did some cleanup and made a v1.1 release with stable version in trunk with 2.4.2p12, I put the beta 1.1 version in trunk with 2.4.2p12  and deleted it from the branch, it look stable and work well even if there is a few build errors like the release v1.1.(variable unused). I updated the readme.md in trunk and will add a version info in the wiki.
Tell me if it look more "clean" now or if something missing  :).
If there is no major issues with new trunk (still testing it..) do you think we can  make a release with it and put 2.4.3b29 in trunk ?

Francklin2, 2017-09-11:
All version in 2.4.2p12 works well on the Pi 2 & 3, so I tryed to build in 2.4.3b29, I took the rtkrcv.c str2str.c rtklib.h vt.c vt.h from 2.4.3b29 and add the modified line from RTKBASE, it finaly builds and work well but I saw a issue in PPP who can't get a fix: now in 2.4.3 we can only use a L1+L2 or L1+L2+L3 in PPP,a single frequency will not work. 
I saw that after testing  original rtkrcv on command line in PPP : same result, then I look on RTKlib issue and in issue #305 there is this post from tomojitakasu :  

> In 2.4.3, PPP always needs dual- or triple- freq observables.
> With L1 only data like u-blox, it does not work.     

https://github.com/tomojitakasu/RTKLIB/issues/305

Francklin2, 2017-09-17:
A branch was uploaded with a version that build now so i can close the issue

closed 2017-09-17 by Francklin2