Francklin2/RTKLIB_Touchscreen_GUI #33: build_rtkbase.sh does not contain all neccessary packages to build RTKBASE

exi2163, 2017-09-21:
The script build_rtkbase.sh does not contain all neccessary packages to be able to build  RTKBASE.
I needed to additionally install qtcreator and qt4-dev-tools to be able to build. 

If the needed packages are not installed qmake will fail and the user will not be notified. Also there is no information on where to find the log which contains the error. The success message could be based on the presence of the RTKBASE binary at the end of the script to prevent confusion.

Francklin2, 2018-02-15:
Hi  Thank you for the report, I think you are installing from a fresh image of raspbian, the script we have was based on a image with qtcreator already installed. We have to do a script for a fresh install on raspbian

Claudio033, 2018-02-15:
Hello,
Congratulation for this application and its evolutions.
As future evolutions, I suggest you integrating it the projections Lambert-93 and Corresponding Conics 9 zones (CC42 to CC50).

jbeilin, 2018-02-18:
Hi, 
What about linking Proj4 to the project. That could be interesting for @ENSG_IGN students. 
We just have to see whether it's enough work for a project.
Jacques

Francklin2, 2018-02-20:
Hi Claudio
You mean I have to add the projection options in the setting tabs ?

Francklin2, 2018-02-20:
For the linking with proj4, it's a good idea but I dont know how to do it, I will take a look on linking this week

kikislater, 2018-02-20:
Proj.4 is a lib in common GIS software. Inside the lib there is a command line program which can convert data from one projection to another. It's very powerfull. Program name is cs2cs and usage is like that : 
`echo "-1.176492 45.587111 " | cs2cs +init=epsg:4326 +to +init=epsg:2154`
This command convert from Lat/Lon to Lambert 93. There is python binding for this command and proj4js, a javascript implementation to enhance your web script... 

Proj4 is available with this package under rpi : libgeo-proj4-perl and does not need rebuilding it. So usage is just to call cs2cs or use python-proj4 or use proj4js
source : http://micmac.ensg.eu/index.php/Compilation_de_MicMac_sur_ARM_(RaspBerryPi)

Francklin2, 2018-02-20:
I see so Claudio want a tool to convert results in another projection (I thought it was the projection option in rtklib). cs2cs from proj4 could be integrated in rtkbase to do that and write the output to a file and the screen

Francklin2, 2018-04-13:
I did a few test with cs2cs and was able to use it as a cmd line in rtkbase to convert the post processing results from rnx2rtkp in ECEF. so I used this command line to convert to WGS 84:
cs2cs +proj=geocent +datum=WGS84 +to +proj=latlong +datum=WGS84 -f" "%.12f"
I'm not sure it's the correct way to do that, do the XYZ(ECEF wgs84?) results from rnx2rtkp are correctly described in " +proj=geocent +datum=WGS84 " . On the output side Do i need more arguments for the wgs84 or is it better  and safe to use the EPSG ? :
cs2cs +init=epsg:4328 +to +init=epsg:2154 (for geocent WGS84 to Lambert93)
I can add in the interface the choice of EPSG and have a output in a separate file

Francklin2, 2018-04-15:
I made a update with cs2cs, in the save point option, you can choose Lambert93 and the CC42 to CC50, the converted coord will be added at end of saved point in pointfiles foldes.
cs2cs or proj5 have to be installed before on the system (It stil have to be add to the build script)

kikislater, 2018-04-16:
Hi Francklin, 
Great addition.

WGS84 is 432**6** and not 4328. 4326 is more standard and it's the output of rtklib
http://spatialreference.org/ref/epsg/4326/
https://en.wikipedia.org/wiki/World_Geodetic_System
4328 is variant. Don't use it here ! 

Next step would be geoid correction ...

Francklin2, 2018-04-22:
I add a new build script for RPI3B+ and Qt5 or test, it build but the post process part still not working (Qlist out of range) I put this in the  dependencies / apt get part
proj-bin qtdeclarative5-dev qttools5-dev-tools libgles2-mesa libgles2-mesa-dev build-essential qt5-default
Code is working when build on X86 desktop PC and qtcreator on qt 5.7.1 ...

Francklin2, 2018-04-28:
I done a update and found why it was not working on RPI3, the RNXCMP library (Hatanaka compress) in package were build for x86...I add a bin-arm and bin-x86 folder with compiled binaries.
Now cs2cs is used in post process to convert XYZ to LLH with :  
cs2cs +init=epsg:4978 +to +init=epsg:4326 -f %.6f 
Is the 4978 ok for xyz or should I put back this syntax +proj=geocent +datum=WGS84  
The convbin and rnx2rtkp lib are not included in the scripts and the rights of binaries in package folder have to be checked too. 
By default I put the ARM binarie for Hatanaka in package folder. I also got a Qt'4 version that should work now (will test it soon on a raspberry)

Francklin2, 2018-04-29:
I added 2 scripts: 
- build_rtkbase_Qt5.sh  ,works on a raspbian stretch and RPI3 (not tested on RPI2)
- make_library_process.sh in lib/rtklib to build convbin and rnx2rtkp , this script is called by build_rtkbase_Qt5.sh

Francklin2, 2018-05-11:
@kikislater and claudio033 
I add the geoid correction file option to the cs2cs transformation and opened a new issue for Proj4/cs2cs use in rtkbase .