Francklin2/RTKLIB_Touchscreen_GUI #34: Autostart

KlausiKleff, 2018-01-12:
Hello,
I like this project. It work very well !
I have an question:

Is it possible to make an autostart (e.g. Rover RTK Mode) when starting the raspberry ?

Francklin2, 2018-02-15:
Hello
Thank's for the report and the idea of autostart, I keep that for the next development, we can do the update for this summer.

Francklin2, 2018-05-08:
Hello 
I added a autostart option in master branch and in a new RTKbase_QT4 branch (for RPI2 and older images under QT4). On main screen you have a autostart options button, in the option page you can enable autostart at boot and choose rover mode (base mode still need to click on start to run) you can choose one of the 6 default mode or custom 1 from custom config.
On each boot it will run desired rover mode, To disable autostart, stop calculation and set OFF in autostart option to run RTKbase as usual