Francklin2/RTKLIB_Touchscreen_GUI #35: RTKLIB_touchscreen GUI compatible with Raspignss?

DriverKit, 2018-03-20:
I would like to use RTKLIB touchscreen with a Raspignss but I am not sure if it will compatible?  
Where could I find a basic getting started guide for RTKLIB touchscreen (preferably in English).

Great project - looking forward to using it!

Thanks for your help

Francklin2, 2018-03-22:
Hello
For a basic startup you can take a look in the wkiki here :+1: https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/wiki
To use the Raspignss with rtklib and rtkbase you can take a look about NVS08C and rtklib, this video show that it should work
https://www.youtube.com/watch?v=3het9GEIwkI
In gnss protocol setting you have to set NVS instead UBX

Best regards
Francklin

RRIoT, 2020-06-01:
Hi.
The I also have a RasPiGNSS.
The webpage link that is on the Description of the YouTube has been taken down
Does anyone have that info that was supplied on that link.
Thank you for your help.

Francklin2, 2020-06-03:
Hi , The youtube link is working for me, I saw also a new version of the vidéo here
https://www.youtube.com/watch?v=0gesSi-OIJU

You can also try this:
- Install a new raspbian
- Follow the installation steps for raspiGNSS from here:
https://drfasching.com/products/gnss/raspignss-aldebaran/installation/
- Once your raspiGNSS is configured and tested, note the connection setting (tty AMA0, parity etc), then install rtkbase with the install script and set the connection with the connection setting you have noted and NVS instead of UBX  for the stream.

RRIoT, 2020-06-05:
Hi Francklin2

Thank you for your quick reply and Awesome Software.

The link I was talking about, was not the YouTube Links themselves, but the Link in there Description.

The Link is https://pixhawk.org/dev/rtk-basestation-touch-tft.

I believe this has the circuit Diagram, etc for the NV08c's, that is in the video.

Kind Regards, Craig

Francklin2, 2020-06-18:
Hello

I didnt find the vidéo too, wich NV08c board do you have ?  a raspignss or another brand ?I can help you for the wiring on  a pi. You should only plug the board power (5V or 3.3V) and the serial TX and RX (TX and RX have to be crossed; TX plug to RX and RX to TX) You can find the pinout of the raspignss here
https://drfasching.com/products/gnss/raspignss-aldebaran/raspignss-specifications/

Kind regards

Francklin