Francklin2/RTKLIB_Touchscreen_GUI #36: Componenti per il progetto RTKLIB

cboscaro, 2018-03-22:
Good morning.
I wanted to ask if you can give me an updated list of the best components (Hardware) to buy to carry out the project.
 
Best regards

Francklin2, 2018-03-22:
Hello
I check the links on the readme.md (end of first page) for the components and they all seemed to work, they are still up to date  but you should take a raspberry pi 3 instead of the raspberry î 2 to be compatible with the new post processing version

Best regards 
Francklin

cboscaro, 2018-03-22:
Thanks Francklim.
I installed it on raspberry 3 and it works fine. I wanted to know if the GPS antenna and other components are the best. 
Thank you.
Cristian

foxittt, 2018-03-30:
Is it possible  to obtain different corrections via USB 3G modem?

Francklin2, 2018-03-30:
Yes if you have a internet connexion via 3G  you can acces to correction, just make sure that you 3G usb is compatible with linux and you will have to install drivers and utility for it

foxittt, 2018-04-04:
Is Pré-installed image of RTKbase with March 2018 update compatible with Pi3?

Francklin2, 2018-04-07:
For the moment there is only one pre-installed image  for RP3 here :+1: https://drive.google.com/open?id=0B5xmVAi4jATOcEQ1TWxaOXprMk0
but the version is before the 2018 updates so you will have to update the sources files from the github and perhaps update QT4 to QT5 if it doesn't compile, I will try to update the RPI3 images for nxt month

Francklin2, 2018-04-17:
I'm working on a new image but you can try this:
- Follow this tuto to install qtrpi http://www.qtrpi.com/faq#howtoinstall
-Install cs2cs with sudo apt-get install proj-bin
-got to RTKbase folder and do:
make clean
qmake
make -j4

Francklin2, 2018-04-20:
I also have some problem with the new RPI3B+ : the Jessie distrib doesnt boot on it and the new stretch is booting on the RPI3B+ but installing QT5 on it is not easy, if I do  on a fresh image ,: sudo apt-get install qtcreator qt5-default I can compile but the part of rtkbase about post processing wrote in qt5 is not working (it's ok on desktop).

foxittt, 2018-04-23:
The utility for USB 3G modem wants admiistrative password.
Can you write it?

Francklin2, 2018-04-28:
Wich utility is it ? I saw there is differents one depending on USB 3G dongle model.
 I updated the spurces and it should build on rpi3 now , so I can start to work on new image now

cg-david, 2018-05-03:
Small feedback from here. We used the GoogleDrive link in [the wiki](https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/wiki) for a Raspi 3. Booted flawlessly, boot screen came up and everything worked fine.
We have two _ublox C84-M8P-3_ and want to use one as base, the other as rover. My student will connect the _ublox_ to the Raspbi tomorrow and see how far she gets. @Francklin2 would you think your build would work out of the box for this setup, or do we need to install / compile more?
We just need a decent and stable build, not the very latest.

Francklin2, 2018-05-04:
Hello
The image for the rpi3 should have the last stable version, you don't have to compile more things on it. Just check if your Ublox M8T is well configured , you can follow the wiki for that part https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/wiki/6-Setting-the-Ublox-M8T-for-RAW
For the link between rover and base you may have to use the custom presets if the default configuration dont transmit data

foxittt, 2018-05-05:
Hello
How to config RTKbase for NV08C-CSM?

Francklin2, 2018-05-08:
In gnss protocol setting you have to set NVS instead UBX (you can try with custom config or change the conf files)

cg-david, 2018-05-08:
@Francklin2 We use Ublox M8**P**. Will the setup be the same?
We also did a first test with ublox-center under windows. The base station did not receive a FIX, but the rover did after some time. Even in FLOAT, the rover was quite accurate (below decimeter) in lat/lon and okay (decimeter) in height. Does the base station need to be in RTK-FIX-mode?

foxittt, 2018-05-08:
As I know, NV08C requies some command after starting for raw data sending. How can I send theese commands to receiver?

Francklin2, 2018-05-08:
You have to edit the .conf file and put the path of your cmd file in file-cmdfile1 field
You  can take a look here to see about rtklib and nv08c  http://lists.osgeo.org/pipermail/foss-gps/2015-September/001448.html

Francklin2, 2018-05-08:
@cg-david Thank's for the report about your results(sub decimeter is very good!!), I think the M8P should work out the box (I think SFRBX is enabled) .If you have to find the base position, you can try the post processing to log raw and get base position by post processing.

Francklin2, 2018-05-10:
You have to change these 2 lines in your conf files:
inpstr1-format     =nvs        # (0:rtcm2,1:rtcm3,2:oem4,3:oem3,4:ubx,5:ss2,6:hemis,7:skytraq,8:gw10,9:javad,10:nvs,11:binex,12:rt17,15:sp3)
file-cmdfile1      =../../../data/nvs_raw_10hz.txt

then create a data folder in RTKBASE , download the zip file from there : http://www.onetalent-gnss.com/ideas/usb-hw-receivers/denga10/rtklib242b4_mb.zip?attredirects=0&d=1
 and copy the 2 nvs_raw_1hz.txt files in the data folder

Here is a conf file for nvs

[RTK-Kinematic.zip](https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/files/1993050/RTK-Kinematic.zip)

foxittt, 2018-05-14:
@Francklin2
Unfortunately, nothing happens.
RTKBASE persistently does not see the stream from the receiver. I do not know where to look anymore.

Francklin2, 2018-05-14:
Ok to do faster tests we can try first with rtknavi under windows to see if the NV08C got everything to run rtklib. after you can look at the rtknavi conf file to see what was the good setting. check also wich port is used by the NV08C with ls /dev/tty* (can be AMA0 or USB0) . try to put the abssolut path to the cmd file

foxittt, 2018-06-05:
Hello. 
Is there news about the new version of RTKBase and the use of NV0C-CSM receivers?