Francklin2/RTKLIB_Touchscreen_GUI #39: getting a signal but no sync to satellites

gisma, 2018-05-14:
Hi @Francklin2 
first thanks for the nice project. Great idea. I just try to adapt it for my my courses...
I've  setup  and assembled a copy of your hardware the only difference to your proposed setup is that I did not weld the cables but use the usb port to connect the GNSS Ublox M8T to the raspi 3.  I have installed your image . So finally I am obvously getting a signal from the antenna but there is no sync to the satellites  or by other words the signal is not decoded. What do I miss?
Any suggestions are welcome. 
cheers Chris

Francklin2, 2018-05-14:
Hello
Did you set the ublox M8T to send raw data ? you can follow this wiki to check the M8T setup:
https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/wiki/6-Setting-the-Ublox-M8T-for-RAW
ypu can also test the M8T with rtknavi under windows

gisma, 2018-05-15:
Hi 
I actually have set all params as you described int your tut. While checking them again I noticed that I am not abnle to set the PRT UART1 in protocol to 0+1+5 UBX+NMEA+RTCM3.when sending it to the MT8 it is always reset to 0+1 UVX+NMEA

gisma, 2018-05-15:
Well using the u-center software I am retrieving the satellites and all data. Back to RTKBASE I see just bytes coming in but no satellites are retrieved. I am using the preinstalled image with an Raspi 3b+ any needs to change libraries or update issues?

gisma, 2018-05-16:
After setting up the raspi from scratch and building the current master branch from scratch (had to install some qt dependencies It finally runs. Thanks for support!
cheers Chris

closed 2018-05-16 by gisma