Francklin2/RTKLIB_Touchscreen_GUI #41: Describe how solution from RGP or EUREF is compute

kikislater, 2018-08-06:
Salut @Francklin2,
Tu peux décrire la méthode pour le calcul depuis plusieurs corrections stp ? 
- Est-ce le fait de calculer plusieurs solutions puis d'en estimer la moyenne des différentes corrections ? 
- Ou est-ce inclut dans un process rtklib que j'ai manqué ?
 
Merci

Francklin2, 2018-08-22:
Salut 
En fait c'est une fonction dans RTKbase, je ne l'ai pas vu dans RTKlib,  dans RTKbase on fait plusieurs calculs en utilisant a chaque calcul une station RGP differente, ensuite on en fait une position moyenne, pour cela il y a eu 2 versions du calcul de la position moyenne:
- la version d'origine  ( https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/tree/Post-Processing ) ou on prends les 2 premiers points, on en calcule le point médian puis de ce point on refait un point médian en prennant le point suivant. le calcul se trouve ici  en bas de page : https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/blob/Post-Processing/RTKBASE/rnx2rtkp.cpp
Je n'ai pas pu le refaire en variant le nombre de calculs et sous QT4 je ne retrouve pas les fonctions de math utilisées .
- l'autre méthode est de faire la moyenne des valeurs X Y Z , j'ai gardé celle la car plus simple a manipuler mais je ne connais pas la diffrence de précision entre ces 2 methodes ..

kikislater, 2018-09-23:
Ok merci Francklin ! 
Par contre, ça ne fait pas le calcul sur le fix (Q = 1) uniquement ... Du coup tu moyennes du float, fix, je penses que ça peut-être amélioré pour n'avoir que la moyenne sur le fix

Francklin2, 2019-04-16:
Salut Kikislater
Effectivement jr viens de revoir le fichier .conf utilisé pour le post processing qui se trouve dans le dossier "work", j'ai vu qu'il est configuré pour du L1+L2 uniquement sur le GPS (pas Glonass).
Pour le calcul uniquement sur le fix tu sait quel réglage il faut changer dans le fichier .conf ?
PS: sur la version 2.43 le post processing et rnx2rtkp ne fonctionne pas sur mon pi mais cela fonction sur PC/ubuntu, j'essaie de voir si il ne manque pas un truc dans les librairies et le build script)