Francklin2/RTKLIB_Touchscreen_GUI #42: Missing RTKBASE

carlosmry, 2018-08-18:
Hi!

I have been following all the installation steps on a Raspberry pi 3B+. ./build_rtkbase.sh seems to build everything fine, but there is no RTKBASE in the folder that I can run (./RTKBASE). Any ideas why this might be happening? 

Thanks!

robinsjanis, 2018-10-07:
Hello!
I tried to run the build script two times, and the command ./RTKBASE does not work - no such file or directory. What am I doing wrong?

mwruschel, 2019-01-08:
Good afternoon, I'm having the same problem with my Raspberry pi 3 using the raspbian-stretch image 2017-09-07

when executing ./build_rtkbase.sh it finishes. but when I execute the ./RTKBASE it says that the truck does not exist ...

Could Francklin2 help me?

MariusBredesen, 2019-01-17:
I got the same issue.  ./RTKBASE does not run. I also tried downloading the img with RTKbase pre-installed, and this wont boot at all :/  Please let me know if any1 got a solution.

robinsjanis, 2019-01-17:
I tried a lot of things and gave up on this project.
I tried this https://github.com/taroz/TouchRTKStation,
I could boot it to the GUI, RasPi3 recognized the Ublox M8T reciever, but the program couldn't post the signal to NTRIP server. So now my Raspberry is in a drawer waiting for a better days to come.
I will probably use old netbook with Rtklib for the base. 
If anybody comes up with a solution for this RasPi based rtk base station to work, please, email me robinsjanis@gmail.com

gisma, 2019-01-17:
Same with me. Great idea and project. 
However it would be very helpful and highly  appreciated if the build process could be revised  or an actualized version of the pre-build could be provided.
cheers Chris

MariusBredesen, 2019-01-17:
> I tried a lot of things and gave up on this project.
> I tried this https://github.com/taroz/TouchRTKStation,
> I could boot it to the GUI, RasPi3 recognized the Ublox M8T reciever, but the program couldn't post the signal to NTRIP server. So now my Raspberry is in a drawer waiting for a better days to come.
> I will probably use old netbook with Rtklib for the base.
> If anybody comes up with a solution for this RasPi based rtk base station to work, please, email me [robinsjanis@gmail.com](mailto:robinsjanis@gmail.com)

I will try the taroz version. I got the Ublox ZED-F9P, i will let you know if i get it to work.

Francklin2, 2019-01-17:
Heelo
Sorry for the lack of updates and news on the project, I still have some problems to do a clean install script with raspbian strech for pi 3, I was able to make a build on raspbian strech on a Pi3. You can try the following steps:
- get a fresh imafe of raspbian strech, install it on a SD card
- do a sudo apt-get update and a sudo apt-get upgrade
- clone the Rtkbase project from the github
- update rights of build_rtkbase_Qt5_ARM.sh in RTKBASE folder and lauch it
- at first lach it will hang on RTKBASE compilation, unplug and reboot, then launch again build_rtkbase_Qt5_ARM.sh this time it should give a working RTKBASE executable
-install screen driver from waveshare
- do a autoboot for Rtkbase, I got dsome problems on this part with raspbian strech and have to use the autostart file 
- I was not able to get a splash  screen in raspbian strech, perhaps I have to try again on a clean raspbian image...

I made a new preinstalled image for Pi 3 with rapsbian strech, it's for a 16Gb  SD cardand the splash screen is not working yet, I will try again for a 8Gb image.
https://drive.google.com/open?id=1A0_4Cfmd_YapDnDZIPAidAfNqGR9QZ-S

We also tried the Taroz project and it working well and gives good results, Our project for this year will be to do a fork of the Taroz project and add some fonctions of Rtkbase to the Taroz project like the post processing or more configuration option, this should be done around may/june.
If you install the Taros project on a RTKNASE hardware, the only diffrence is the screen size, so in the install build sript you may have to change the screen size from 4inch to 5inch (use LCD5-show instead LCD4-show)

Francklin2, 2019-01-29:
Hello
There is a new preinstalled image for PI3 based on raspbian stretch, this is for a 16 Gb SD card, you can find it here : 
https://drive.google.com/open?id=1HOssKfLAgLO0DKe1ZVBqeMXHCmyP4w-P

mwruschel, 2019-02-01:
Good morning Francklin, I'm developing a project with rtklib. But I do not have a good knowledge in development.
I would like to know if you have no interest in building with me commercially, and receive for your service ...

You can send us an e-mail so we can talk better, thank you.

mwruschel@hotmail.com

Francklin2, 2019-02-03:
Hi Foxit
On this new image version, it was based on latest raspbian strech from nov 2018, it was a 16GB image, the one I realesed last week was made with a 32Gb image, the rtkbase software is the same (version 2018 with post processing).
I made a new build script (build_rtkbase_Qt5_ARM.sh) and some install scripts, you can find them here:
wget https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/raw/master/RTKbase_installer.zip
Unzip the installer file in home directory (/home/pi)
Go to the folder /home/pi/Installer
Update rights for the 3 scriptswith chmod +x 1-Download_rtkbase.sh chmod +x 2-Build_rtkbase.sh chmod +x 4-Install_LCD_driver.sh
Lauch the 3 scripts in order to 1 downmload and build RTKbase 2 Install autostart and 3 install driver

@mwruschel I send you a mail this evening

kikislater, 2020-01-20:
project moved here : https://github.com/Francklin2/EasyGNSS

closed 2020-01-20 by kikislater