Francklin2/RTKLIB_Touchscreen_GUI #43: NTRIP connection with username and password

MariusBredesen, 2019-02-14:
Hello, this is not really an issue, but if someone is able to help me out that would be amazing.
I am currently trying to connect to a NTRIP client to recieve correction data for my ZED-F9P. But im not able to figure out how to enter my username and password correctly in the string to authenticate correctly. Maybe someone could post an example string with a login. 
Thanks in advance.

Francklin2, 2019-02-14:
Hello
In the edit configurationgo to page input in input stream 3 select NTRIP and in the stream path, you have to set the line like this:
login:password@www.euref-ip.net:2101/BRST0 
so before @ you have the login and password, and after @ the server adress (www.euref-ip.net), his port (:2101), and the station name (/BRST0)

robinsjanis, 2019-02-24:
Hello!
Speaking about NTRIP.
Is it possible to send rtcm3 data over ntrip to rtk2go.com? I can't find such option in settings

Francklin2, 2019-02-25:
The output part for str2str and rtk2go.com should look like this:
ntrips://:BETATEST@rtk2go.com:2101/yourstationname:RTCM3

The rtklib manual give these option for the stream out
stream path
 serial : serial://port[:brate[:bsize[:parity[:stopb[:fctr]]]]]
 tcp server : tcpsvr://:port
 tcp client : tcpcli://addr[:port]
 ntrip client : ntrip://[user[:passwd]@]addr[:port][/mntpnt]
 ntrip server : ntrips://[:passwd@]addr[:port][/mntpnt[:str]] (only out)
 file 

Perhaps you will have to ask for the latest password for rtk2go
http://www.rtk2go.com/how-to-connect/
Are you  able to send data to rtk2go with the windows version of str2str ?(strsvr.exe)
I have to see if it's possible to input the path in rtkbase option

MariusBredesen, 2019-02-26:
Hello again, i got a few more questions:
Is it possible to make RTKbase display ETRS89 coordinates?
Also, would it be possible to get your email adress? im working on a bachelor thesis and will probably have alot more questions in the next months. my email is: marbred@stud.ntnu.no

Francklin2, 2019-02-26:
If you look in the .ini file in the windows app you will find the correct way to write your ntrip client access, it should look like this syntax :
login:[password@www.euref-ip.net](mailto:password@www.euref-ip.net):2101/BRST0
If you do a post processing you can use a EPSG projectionif the EPSG4258 can be used instead ERTS89 , you can use it in post processing (I have to look if it's possible to enter manually the EPSGxxxx).
I made a update for the ntrip server, so in the base mode at the out strem option, you can choose ntrips:// and enter the right string in the server adress box.
You have to update the sofware for that, it's only for the RPI3 and QT5 for the moment, if you have a RPI2, I have to update the QT4 version. To upgrade a Raspberry pi3 follow these steps:
- delete or rename rtklib touchscreen gui folder
- Download install scripts with: wget https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/raw/master/RTKbase_installer.zip
- Unzip the installer file in directory (/home/pi)
- Go to the folder /home/pi/Installer
- Update rights for the 3 scriptswith chmod +x 1-Download_rtkbase.sh chmod +x 2-Build_rtkbase.sh chmod +x 4-Install_LCD_driver.sh  
- Execute only Download_rtkbase.sh
I'll send you a mail keep contact for later

MariusBredesen, 2019-02-27:
I checked the .ini file, and the string ive entered in RTKbase is correct. i connect to the NTRIP client, but i get timed out after about 10-15 seconds due to the ntrip client not recieveing ant NMEA string from my rover. (the ntrip client require a NMEA GGA string every 10 seconds to locate the closest basestation)

- https://gyazo.com/ed4d635068b6d82d50f5cd6f92893206  this is the NMEA option i get on the windows version, if i turn this off, i get the same same timeout error as i do on the rpi3b+. is there a similar setting on RTKbase im not finding? or is the RTKbase-rtklib version too old?

- https://gyazo.com/a522896a1e7c0e453c8115e3f9a35476 and this is what the timeout message looks like on RTKbase.

Francklin2, 2019-03-02:
Hi I made a new update and now the ntrip server string is correct, tested on rtk2go with this parameter:
:BETATEST@rtk2go.com:2101/Myplace#rtcm3 (it was :BETATEST@rtk2go.com/Myplace;2101#rtcm3)
For the ntrip client timeout error you have, can you send the input parameter inpstr2-path  ?
if you have 2 RTKbase they need to have the same version of rtkbase

MariusBredesen, 2019-03-05:
I can send you the actual string on email, but i havent gotten any email from you yet. If you tried sending me a email on the previous adress i posted there might be a spam problem because its my school email.
It might be better to email me at marius.bredesen@gmail.com

Francklin2, 2019-03-30:
Hi marius 
I have seen in the server adress that the mountpoint is ending by : 
HREFNN2000:
it should end like this
HREFNN2000
I will make a try this weekend with this adress to see if the connexion works, with the zed-f9p you have to use the 2.43 branch including rtklib 2.43b31.

Francklin2, 2019-03-31:
The timeout was caused by the option inpstr2-nmeareq     =off in the conf file, it have to be set to single to transmit NMEA to the VRS station. In the 2.43 branch added this option in the edit config in MISC. After a few minutes you will have a single solution the the error should be gone and you will see some bytes transmited to the base (bytes inputed are not readed)
Tell me if it works now with your zed-f9P and the NTRIP base station

MariusBredesen, 2019-03-31:
Alright, thanks! Will try this out today! Will update you later :)

MariusBredesen, 2019-03-31:
I just tried it out, and it seems to be working perfectly! Thanks alot for
your work, Francklin! I will be testing it out more tomorrow, and will
compare results with Leica GS16 rovers.

Attached a pic of the status screen with Fix :)

søndag 31. mars 2019 skrev Marius Bredesen <marius.bredesen@gmail.com>:

kikislater, 2020-01-20:
project moved here : https://github.com/Francklin2/EasyGNSS

closed 2020-01-20 by kikislater