Francklin2/RTKLIB_Touchscreen_GUI #45: Installing 2-Build_rtkbase.sh

ajshaww, 2019-03-18:
On a Rpi3 fresh install of rasbian I get the error "-bash: ./2-Build_rtkbase.sh: /bin/sh^M: bad interpreter no such file or directory". After googling I found that the command "sed -i -e 's/\r$//' 2-Build_rtkbase.sh" fixed the issue and then it runs fine after that.

kikislater, 2019-03-18:
Ok thank you bad line ending in the file and header not present ! 
@Francklin2 I modified the file rtkbase.sh but where is 2-Build_rtkbase.sh ? In RTKbase_installer.zip I guess ? Could you put it uncompressed on master ? Github has release functionality to put zip or tar.gz or other ...

Francklin2, 2019-03-18:
Hello
I was stuck with the issue " missing rtkbase" after build so I made a dirty but working solution, I séparate the script in 2 parts, build the library  then call the build script for rtkbase 2 times (I didn't found why it doens't build on the first try but build on second try). So now you have one script to build on ARM processor and one for desktop x86 :
build_rtkbase_Qt5_x86.sh, they worked on a raspbian stretch and a pi 3 (and after sudo apt-get update and sudo apt-get upgrade) 
build_rtkbase_Qt5_ARM.sh
 The install package is in the root of the github master here:
https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/blob/master/RTKbase_installer.zip
I can also put them in a folder on master, I put in these scripts:
- clone github in home directory and build rtkbase 
- download and install touchscreen drivers
- create autostart rtkbase at boot

Francklin2, 2019-03-26:
Hello
I put the unziped build scripts in a Installer folder, so we can update the files now. 
@kikislater 
It's possible that I erase the change you made in RTKBASE.sh, can you check if the change in the missing header are still there ? (I think I did a bad commit/update..)

kikislater, 2019-03-26:
Dear Francklin,
Your latest commit is here : https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/commit/660ae8590d20ab6d1be923b0310e58f7840fdc48

So no modification in RTKBASE.sh
So what it's not good : 
1-Download_rtkbase.sh : doesn't execute the script
2-Build_rtkbase.sh : contain lines present in 1-Download... and execute only script for ARM.

With case in bash, it's possible to execute script according to computer specification : ARM or x86. Is this x86_64 ?

These files need refactorization !

Francklin2, 2019-03-26:
Thanks,  I recheck the commits to see the changes, I will remove the same chmod present in the two scripts, I also need to find something for the differents home path on pi and linux.It can be a issue if we try to download on linux. For refactorisation I will try this on build script : 
```

 `
#!/bin/sh

# Build RTKBASE 
cd ~/RTKLIB_Touchscreen_GUI/RTKBASE

PS3='Please enter your choice: '
options=("Build on ARM processor(Raspberry pi)" "Build on x86 processor(desktop PC)" "Quit")
select opt in "${options[@]}"
do
    case $opt in
        "Build on ARM processor(Raspberry pi)")
            echo "you choose ARM build"
			./build_rtkbase_Qt5_ARM.sh
            ;;
        "Build on x86 processor(desktop PC)")
            echo "you choose x86 build"
			./build_rtkbase_Qt5_x86.sh
            ;;
        "Quit")
            break
            ;;
        *) echo "invalid option $REPLY";;
    esac
done` 
```

kikislater, 2019-03-27:
In bash $HOME is variable and select /home/$USER where $USER is also a variable.
So just let do for rpi and linux : 
`cd $HOME`

For script you could select like you did but also use native kernel command : uname -m
So in that case it remove user interaction but need more description (about platform specific material like tons of arm) and possible errors ... 

```
# Find current arch
UNAME=$(uname -m)

if [[ "$UNAME" == *"x86_64"* ]] ; then
	echo "Linux x86_64" | ./build_rtkbase_Qt5_$UNAME.sh
elif [[ "$UNAME" == *"arm"* ]] ; then
	echo "ARM | ./build_rtkbase_Qt5_ARM.sh"
elif [[ "$UNAME" == *"i*86"* ]] ; then
	echo "x86 32bits" | ./build_rtkbase_Qt5_x86_32.sh
fi

```

Just precise if this is x86_64 or not and may be rename bash script with this information. 32 bits are commonly named i386 or i686, 64 bits x86_64

You could include also in one script and make check it by gcc : 
https://stackoverflow.com/questions/34747838/how-to-tell-if-program-is-running-on-x86-x64-or-arm-linux-platforms

Francklin2, 2019-03-27:
I remove the chmod in the download script and use $Home in path, so it works on raspberry and linux x86 PC user name, a menu was added in the build script to choose ARM or x86 build, it should work better. I can try to merge all the script with a menu to have only one script to download.

Francklin2, 2019-03-28:
Thank's for the platform detection srcipt, I put in the build script and rename build_rtkbase_Qt5_x86.sh to build_rtkbase_Qt5_x86_64.sh and add build_rtkbase_Qt5_x86_32.sh (same as the x86_64).
I tried to merge all the script in one like this, but i have to try it on a pi
`#!/bin/bash

# Install RTKBASE 

while true
do

PS3='Please enter your choice:'
options=("Download RTKbase" "Build RTKbase" "Install autostart at boot" "Install 5 inch LCD driver" "Quit")
select opt in "${options[@]}"
do
    case $opt in
        "Download RTKbase")
            echo "Install RTKBASE folder from github"
            cd $HOME
            git clone https://github.com/Francklin2/RTKLIB_Touchscreen_GUI.git
            # restore rights for build script
            cd $HOME/RTKLIB_Touchscreen_GUI/RTKBASE
            chmod +x build_rtkbase_Qt5_ARM.sh
	    chmod +x build_rtkbase_Qt5_x86_64.sh
	    chmod +x build_rtkbase_Qt5_x86_32.sh
	    break
            ;;	
        "Build RTKbase")
            echo "Starting RTKbase build"
	    cd $HOME/RTKLIB_Touchscreen_GUI/RTKBASE
            # Find current arch
	    UNAME=$(uname -m)
	    if [[ "$UNAME" == *"x86_64"* ]] ; then
	    echo "Build for Linux 64 bits" | ./build_rtkbase_Qt5_${UNAME}.sh
	    elif [[ "$UNAME" == *"arm"* ]] ; then
	    echo "Build for ARM | ./build_rtkbase_Qt5_ARM.sh"
	    elif [[ "$UNAME" == *"i"* ]] ; then
	    echo "Build for Linux 32 bits" | ./build_rtkbase_Qt5_x86_32.sh
	    fi
            break
            ;;
        "Install autostart at boot")
            echo "Installing autostart"
            mv RTKBASE.desktop ./.config/autostart/
            cd $HOME
            mkdir .config/autostart
            chmod +x ./.config/autostart/RTKBASE.desktop
            break
            ;;
	 "Install 5 inch LCD driver")
           echo "Installing LCD driver"
	   cd $HOME
	   git clone https://github.com/waveshare/LCD-show.git
	   cd ./LCD-show/
	   chmod +x LCD5-show
	   ./LCD5-show 
           break
           ;;
        "Quit")
            exit
            ;;
        *) echo "invalid option $REPLY";;
    esac
done
done
`

Francklin2, 2019-04-03:
Hello
I tested on a RPI3 and raspian stretch the build scripts and the install script with menu install_all.sh, it workes exept a build_RTKBASE.sh issue on RPI : on first build it crash and you have to restart the RPI, on second try it build correctly, I saw this issue when we start to build on qt5 instead of qt4, that's why I start 2 times build_RTKBASE.sh  in build_rtkbase_Qt5_***.sh.

kikislater, 2019-04-03:
Do you have errors when it fails to build ?

Francklin2, 2019-04-06:
[RTKBASE.log](https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/files/3050940/RTKBASE.log)

This is the error on the first build, I dont have it when I build under qt creator, I think it may be a outdated file not updated when a do a commit, I updated the pro file and remove the makefile and qmake.stach, it remove some errors but not all. I will check again in the code and mainwindows.cpp

kikislater, 2019-04-09:
A lot to debug in project ! 
Could be a memory leak but all files need review. Error could be seen in qtcreator without needed to build it.
[image]

.pro file is not the good one I think, I had to add ui_mainwindow.h nd startup_options button is not present in this file. May be migration from qt4 to qt5 was not done in the right way. I wonder how it could work in that state :-D

Francklin2, 2019-04-09:
I updated the pro file and remove a mainwindows.h wrote twice time, I will try it on the pi to see if it's better. There is still a lot of errors to remove, the memory leak should come from there.
I dont know why it works too :)

Francklin2, 2019-04-10:
I removed some ui_****.h  files that were uploaded while doing a git pull without doing a make clean in the source files, these files caused the mainwindows error and the need of two build to create a makefile and do again the working make clean before the good build.
Now the script build without crashing but still have a lot of errors to debug, iI check again on rpi tonight, if it build correctly I will open new issues for the other errors

closed 2020-01-20 by kikislater