Francklin2/RTKLIB_Touchscreen_GUI #46: Change package

vmotorkin, 2019-03-18:
Please, change RTKLIB-2.4.2p13 package on RTKLIB-2.4.2b31 from  https://github.com/rtklibexplorer/RTKLIB.git for support receiver Ublox ZED-F9P.
If it possible.
Regards.

Francklin2, 2019-03-19:
Hello 
I'm actually trying to update the rtklib library to 2.43 b31 but i'm taking it from the rtklib 2.43b31 branch from there :
 https://github.com/tomojitakasu/RTKLIB/tree/rtklib_2.4.3
It also support the nice ZED-F9P , it's in progress because rtkrcv , str2str, vt.c and vt.h  have to be patched to work correctly with rtkbase. to change to 2.43b31 we have to:
- copy new src and app folder from 2.43b31 to the folder RTKBASE/lib/rtklib
- copy new rtklib.h to RTKBASE/lib/rtklib
- patch rtkrcv and str2str with code from ENSG, vt.h and vt.c need a few line change

I hope it wont be to long to patch it correctly, there is also the next version of rtkbase in python that will work with 2.43b31 it should be realeased in may

Francklin2, 2019-03-25:
I made a update on the 2.43 branch from 2.43b29 to 2.43b31, it should support the Ublox Zed-F9P. I didn'y have this Ublox so I cant test it, can you tell me if it worked for you ?

vmotorkin, 2019-03-26:
With the RTKLIB 2.4.3b3 it works. I haven't tried RTKLIB_Touchscreen_GUI,
I'm waiting for the 5" LCD screen. I'll try and i'll write.

------
Best regards,
Vyacheslav Motorkin
+7 918 4057326

MariusBredesen, 2019-03-28:
> I made a update on the 2.43 branch from 2.43b29 to 2.43b31, it should support the Ublox Zed-F9P. I didn'y have this Ublox so I cant test it, can you tell me if it worked for you ?

I tried using NTRIP on this new branch with my F9P, but im still getting the same timeout error.

vmotorkin, 2019-04-01:
https://github.com/rtklibexplorer/RTKLIB.git
**rtkrcv** and **str2str** - rtklib 2.4.3 b**31**

https://github.com/tomojitakasu/RTKLIB/tree/rtklib_2.4.3
**rtkrcv** and **str2str** - 	rtklib 2.4.3 b**29**

kikislater, 2019-04-01:
> **rtkrcv** and **str2str** - rtklib 2.4.3 b**29** - not support Ublox ZED-F9P

But b31 is released and seems to support F9P : https://github.com/tomojitakasu/RTKLIB/commit/02a5688c2bf5e9713a5af81161589438cd87b6e5#diff-28316237258e4e44387bde128049efd9

MariusBredesen, 2019-04-01:
I can confirm that the current 2.4.3b31 branch is working nicely on the Zed-f9p. Atleast with static rawdatalogging as well as RTK.

Francklin2, 2019-04-03:
Hi vmotorkin
Did you try again with the 2.43b31 branch  of rtkbase ? it seemed to woks with Marius. As the 2.42 is becoming outdated with the new Ublox, I think I will have to put 2.43b31 on master if it's confirm to work properly

MariusBredesen, 2019-04-03:
Its also worth mentioning that you might have to configure the F9P in u-center to enable ubx-messages, because these are disabled by default for some reason.

Francklin2, 2019-04-06:
Yes it looks to be the same procedure that we have to do with the M8T to enable the raw data in the wiki here https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/wiki/6-Setting-the-Ublox-M8T-for-RAW
I will have to uodate the name in the wiki and add it in the installation instruction of readme.md

MariusBredesen, 2019-05-13:
Is it possible to get a branch with the RTKlib version provided by rtklibexplorer that vmotorkin mentioned in the top comment? this one: https://github.com/rtklibexplorer/RTKLIB

kikislater, 2020-01-20:
Project has moved to python here : https://github.com/Francklin2/EasyGNSS

closed 2020-01-20 by kikislater