Francklin2/RTKLIB_Touchscreen_GUI #48: Test in RPI ZERO W

messplay, 2019-04-08:
Well let me share that the first test compile was "sucessfull" but ./RTKLIB is missing I am running a second compile to view if the file appear. In the other hand I compile in ubuntu and was 100% sucessfull. So if someone have some recomendation it will be welcome

Francklin2, 2019-04-08:
Hello
Thank you for the feedback on the RPI zeroW, did you try the 2.43 branch? ( git clone https://github.com/Francklin2/RTKLIB_Touchscreen_GUI.git --branch RTKbase-2.4.3  )
I just updated it with a missing dependencie for post processing (gfortran)

messplay, 2019-04-08:
@Francklin2  I going to test right now, let me check in my last try Zero freeze =(  let me trying again with new branch

messplay, 2019-04-08:
Well in the last try RPI zero take 6 hours to compile. Now I still waiting.

messplay, 2019-04-09:
Well finally build sucesfull and run in RPI zero. Virtual memory exhausted was the clue. Solved. Just need Add a Swap File to run.

kikislater, 2019-04-09:
@Francklin2 it should be the same and why you need to run the script twice !

Francklin2, 2019-04-09:
Nice to hear you find the trick for the pi zero and the lack of memory, there is still a lot of errors from qt4 to qt5 change (ISO 90 error)and also when I updated to 2.43b31(errors in vt.c, rtkrcv andstr2str and rnx2rtkp). We saw a problem  while updating the files when I do the commit, the .pro file was not updated last time (perhaps some other), I will check the file version this evening.

messplay, 2019-04-09:
hi well next step is test with m8p-0 in few days arrive zed f9 ublox chip. Upload log file for reference.
[RTKBASE.log](https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/files/3060765/RTKBASE.log)

messplay, 2019-04-09:
.rover mode tested without any problem

closed 2019-04-09 by messplay