Francklin2/RTKLIB_Touchscreen_GUI #49: Is possible use ZED-F9 corrections?

messplay, 2019-04-14:
Hi people, one question If I have two ZED-F9 is possible to use self corrections of the module I mean use the output RTCM of ZED-F9 in the serial port of this GUI?

Francklin2, 2019-04-16:
Hello Messplay
To output the RTCM data from your Zed-F9P GNSS to the serial port you have to use the base mode, set your Zed-F9P as input stream (setup should be OK by default) and as output stream you put the RTCM and the adress of your output port (onboard serial port is /dev/tty/AMA0, if you use a USB serial port it will be /dev/tty/USB0). You also have to input the coordinate of the base, you can use the manual input if you already know a precise position, else you can use a rover mode, the last position will be set in automatic position mode)

kikislater, 2020-01-20:
project moved here : https://github.com/Francklin2/EasyGNSS

closed 2020-01-20 by kikislater