Francklin2/RTKLIB_Touchscreen_GUI #50: rnx2rtkp 2.43b31 can't get static solution with L1 GNSS

Francklin2, 2019-04-18:
Hello 
I have a big issue on post processing L1 data with rtklib 2.43b31, after checking why rtkbase post processing was crashing with 2.43b31, I found that the issue came from rnx2rtkp wich doen't give solution in static mode (static and ppp-static), the other mode are working properly.
I tested with the same raw  .ubx file from a L1 ublox M8T on a fresh rasbian strech install and on ubuntu x86_64 PC : same result, if you run rnx2rtkp from 2.42 you have a solution and nothing on 2.4.3, ppp static on L1 is officialy not supported on 2.43 with L1 only, it may be the same for post processing in (rtk) static mode.
It have to be tested with a L1/L2 GNSS to see if there is a solution in static mode with automatic post processing from rtkbase or rnx2rtkp, I don't have a L1/L2 GNSS, if someone can do the test or share a 35minutes L1/L2 raw observation file to try if it works. 
Does someone else  have the same issue, or have a idea to solve it?
Can we mix rtklib 2.43b31 and rnx2rtkp 2.42p13 for M8T users and still have a correct solution ?

salvagilena, 2019-04-25:
Hi. I installed the program from the sd copy of rtklib_pi3. How can the files be updated to the  new folders?

Francklin2, 2019-05-06:
Hello
You can try this new script to update version, download and unzip this file in the installer folder of rtkbase:
wget https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/raw/master/RTKbase_installer.zip
run the install.sh and choose update in the menu, 2.43 for Zed-F9P and 2.42 for M8T  (actually post processing only work with 2.42)

salvagilena, 2019-05-06:
Ok. Thanks

Francklin2, 2019-05-25:
I found where was the problem, it was not the  only issue, it was the position of the base that was incorrect in the conf file and always give a Q=0 in static résult. So the post processing works correctly in 2.4.3

closed 2019-05-25 by Francklin2