Francklin2/RTKLIB_Touchscreen_GUI #51: RTK Issues/bugs

MariusBredesen, 2019-05-03:
Hi, after some testing of RTK with the ZED-F9P ive encountered a few issues that would be nice to fix. 

1. When recieveing corrections from NTRIP client, I am only able to achieve fix solution on GPS and GLONASS satellites. the status page tell me that i have 20-25 satellites observed for both rover and base, but in the fix-solution only 8-12 are used, i have an elevation mask on 10 degrees and SNR mask on 20, which should not be a problem. (i think this may be due to the "Integer Ambiguity Resolution" in the setting 2 menu is only available for GPS and GLONASS) It would be nice if Galileo and Beidou could be added.

2. Another minor issue is that when saving points with RTK, the coordinates and solution freeze until the given amount of measurements is done. I did looked into the point-file, and its only the GUI that freeze, we get different coordinates on the points saved, but some are way off because i was moving and lost fix. But the solution status was also freezed, so i had no idea that i didnt have fix.

Francklin2, 2019-05-14:
Hello Marius
Thank's for the report, I will try to see if it's possible to get infos while saving points, the saving loop is in the mainthread so it freeze until we stop saving.
For the settings, after some tests I found thet the conf files have changed a lot in 2.43, most of the settings lines are still there but the order changed, so we can change the default conf files with the 2.43 version (or the conf files from rtklibexplorer), but for the custom filles I will have some work on the configuration editor,  it will be longer to change that..
By default the conf file are for a L1 GNSS, if you have a L1 L2 ZedF9P you will have to manually change the default modes conf file, if you had a nice config with rtknavi you can keep the conf file for rtkbase.

kikislater, 2020-01-20:
project moved here : https://github.com/Francklin2/EasyGNSS

closed 2020-01-20 by kikislater