Francklin2/RTKLIB_Touchscreen_GUI #52: Install Script Rasbian Stretch on Rpi 2 Increase swap

pauluzs, 2019-05-06:
Just finished installing on Rpi 2 v1.2  with latest stretch image. For this you need to temporary increase your swap, as the 433Mb ram(512-60gpu) and 100Mb swap are not sufficient.
Seen swap usage up-to 276 Mb while using the script. After increasing build was successful in about 1h40min.

Increase swap before installing:
pi@raspberrypi:~/Installer $ sudo su -c 'echo "CONF_SWAPSIZE=1024" > /etc/dphys-swapfile'
pi@raspberrypi:~/Installer $ sudo dphys-swapfile setup
want /var/swap=1024MByte
, checking existing: deleting wrong size file (104857600), generating swapfile ... of 1024MBytes
pi@raspberrypi:~/Installer $ 
pi@raspberrypi:~/Installer $ sudo dphys-swapfile swapon
pi@raspberrypi:~/Installer $ 

run installer:
pi@raspberrypi:~/Installer $ ./install.sh  

Restore default swap when done:
pi@raspberrypi:~/Installer $ sudo su -c 'echo "CONF_SWAPSIZE=100" > /etc/dphys-swapfile'
pi@raspberrypi:~/Installer $ sudo dphys-swapfile setup
want /var/swap=100MByte
, checking existing: deleting wrong size file (1073741824), generating swapfile ... of 100MBytes
pi@raspberrypi:~/Installer $ 
pi@raspberrypi:~/Installer $ sudo dphys-swapfile swapon
pi@raspberrypi:~/Installer $

Francklin2, 2019-05-06:
Hello
Thank's for the info, I didn't try to install the latest version on a pi2 so it's a good news to see that it can work, I will add the swap info in the wiki and see if we can add the swap change in the script

kikislater, 2020-01-20:
Swap is definitively not recommended on sd card ...

closed 2020-01-20 by kikislater