Francklin2/RTKLIB_Touchscreen_GUI #53: Screen Resolution compatibility for 3.5 inch screen

Binabh, 2019-05-23:
Is there any way that I can make this run on 3.5 inch screen also. I Navigated through the code repository but no luck at all. Am I missing something?
All help are appreciated. Thank You!

Francklin2, 2019-05-25:
Hello
If your 3,5inch LCD came from waveshare, you have to take a look at the waveshare drivers here:
https://github.com/waveshare/LCD-show
then look which LCD driver to use (like LCD35-HDMI-480x320-show), when you find the good one, go in the installer folder of rtkbase, edit install.sh at the end of the scripts you will see the LCD driver installation, edit lines 132 & 133 with the correct driver file

Binabh, 2019-06-01:
Many many Thanks!

closed 2019-06-01 by Binabh

comedoo, 2019-07-15:
the driver installer ok, but de window screen of  RTKLib still overscreen ...., Am I missing something?
Thank you !!!