Francklin2/RTKLIB_Touchscreen_GUI #57: 3.5 inch window overscreen in RTKLIB

comedoo, 2019-07-15:
the driver installer ok, but de window screen of  RTKLib still overscreen ...., Am I missing something?
Thank you !!!

_Originally posted by @comedoo in https://github.com/Francklin2/RTKLIB_Touchscreen_GUI/issues/53#issuecomment-511441926_

Francklin2, 2019-08-08:
Hello
The RtKlib interface was made for a 800x600 résolution, is it the setting in your 3.5inch screen ?

comedoo, 2019-09-05:
my resolution is setting for 480x320, so how can I work with this resolution without overscreen ? Is possible to create another version for 3.5 inch ?
thank you !!!

kikislater, 2020-01-20:
You need to modify configuration files. Note that the project has moved to python here : 
https://github.com/Francklin2/EasyGNSS

closed 2020-01-20 by kikislater