Francklin2/RTKLIB_Touchscreen_GUI #64: RTKBASE: No such file or directory

crperez-high, 2021-12-27:
After using the installer script `build_rtkbase_Qt5_ARM.sh`, the following message appears at the end:

```
- script started - Mon Dec 27 21:33:37 PST 2021

  > building RTKBASE
  < done - Mon Dec 27 21:33:37 PST 2021
  - script finished - Mon Dec 27 21:33:37 PST 2021
  < done - Mon Dec 27 21:33:37 PST 2021
  - script finished - Mon Dec 27 21:33:37 PST 2021
```
However, when running `./RTKBASE`, the file seems to be missing.

I'm not sure what I have missed, I am installing on a freshly installed Raspbian GNU/Linux 11 (bullseye).

Francklin2, 2022-01-03:
Hello
Wich script did you used for instakll ? you can try install.sh or install all.sh in the installer folder, I couldnt try yet with latest raspberry pi OS too. you also have to go in the rtkbase folder to lauch ./RTKBASE
Cheers 
Francklin

crperez-high, 2022-01-07:
Hi, thank you for the response.

However, I have also tried to use that script `install.sh` and `install_all.sh`, and get the same result. After building RTKBASE, the file still does not exist:
```
cd /home/pi/RTKLIB_Touchscreen_GUI/RTKBASE
./RTKBASE
```